OpenThermal/libseek-thermal #1: Thermal Seek Pro times out on frame request

eutectic, 2017-05-04:
Hello, thanks for making this library.  Tried to get it running on an ubuntu machine with a seek thermal pro and I'm running into a timeout once it tries to frame.   Any ideas?

`Device Count : 4
vendor: 955  product: 9ff
vendor: 1d6b  product: 3
vendor: 289d  product: 11
bConfigurationValue : 1
ctrl_transfer to/from dev(0x41, 0x54, 0x0, 0x0, 1)
SET OPERATION MODE 00ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
SET OPERATION MOD00ctrl_transfer to/from dev(0xc1, 0x4e, 0x0, 0x0, 4)
Response:   1  0  3  0
READ CHIP IDctrl_transfer to/from dev(0xc1, 0x36, 0x0, 0x0, 12)
Response:  15  0 18  0 56  0 27  0 da  0 d0  0
SET FACTORY SETTING FEATURESctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
GET FACTORY SEtTINGSctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 12)
Response:  31 35 31 38 41 31 43 42 43 4e 33 34
SET FIRMWARE INFO FEATURESctrl_transfer to/from dev(0x41, 0x55, 0x0, 0x0, 2)
GET FIRMWARE INFOctrl_transfer to/from dev(0xc1, 0x4e, 0x0, 0x0, 64)
Response:  30 78 44 30 44 41 32 37 35 36 31 38 31 35  0  0 31 35 31 38 41 31 43 42 43 4e 33 34  0  0  0  0 79  5  1  0  0  0  1  0  0  0  1  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 2)
Response:   0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 2)
Response:   1  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0 31 f7 15  0 18  0 56  0 27  0 da  0 d0  0 31 35 31 38 41 31 43 42 43 4e 33 34  0  0  c  4  c  4 24  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 98 3a
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:  56  1  4  1  0  0  4  0 44  1  0  1  e  0 79  5 4d f2 ec fa  0  0  0  0  0  0  0  0  0  0  0  0  3  0  0  0  0 98 c3 45  0 30 5b c5  0  0 b0 41  0  0 40 40  0  0 14 42 84 71 20 c1 56 79  7 45
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0 c8 42 d0 ed c6 41 e4 84 f1 45  0  0 96 43 c0 f3 fe c1 39 10 b3 46 20 98 3c 43 a2 96 30 3f d3  7 90 a4 9f b1 b2 a9  9 15  a 2f d7 9f 89 2b ed 98 96 30 ea 82 2c b6 ee  5 49 b0 26 3a  a b7
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:  66 4f  2 3d a0 11 18 37 3a 30 8a bd bb a7 1e 42 9b e9 62 29 8b 68 c0 ad 26 f7 4b b3 e2 6e b0 2f 2a 75 60 b6 df 85 52 3c 91 20 db 37 53  f 97 bd de 11 bf 42  0 50 fa 45  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
Response:   0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
[... 197 lines cut ...]
Response:   0  0  4  0 44  1  0  1  0  8  0  0 56  1  4  1  2  0  c  0 40  1 f0  0  e  0  5  0  1  0  0  0 ff ff ff ff  0  0  0  0 ff ff ff ff  0  0  0  0 ff ff ff ff  0  0  0  0 ff ff ff ff  1  0  0  0
ctrl_transfer to/from dev(0x41, 0x3e, 0x0, 0x0, 2)
ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
ctrl_transfer to/from dev(0x41, 0x53, 0x0, 0x0, 4)
Asking for 177840 B of data at 0
Error: bulk transfer failed: LIBUSB_ERROR_TIMEOUT
Error: first frame acquisition failed, retry attempt 1
ctrl_transfer to/from dev(0x41, 0x54, 0x0, 0x0, 1)
Error: control transfer failed: LIBUSB_ERROR_PIPE
ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
Error: control transfer failed: LIBUSB_ERROR_PIPE
ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
Error: control transfer failed: LIBUSB_ERROR_PIPE
ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
Error: control transfer failed: LIBUSB_ERROR_PIPE
ctrl_transfer to/from dev(0x41, 0x54, 0x0, 0x0, 1)
Error: control transfer failed: LIBUSB_ERROR_PIPE
Error: init_cam failed
failed to open seek cam
ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
Error: control transfer failed: LIBUSB_ERROR_PIPE
ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
Error: control transfer failed: LIBUSB_ERROR_PIPE
ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
Error: control transfer failed: LIBUSB_ERROR_PIPE`

maartenvds, 2017-05-05:
Hi, thanks for trying. Will take a look at it.

maartenvds, 2017-05-16:
Sorry for the delay, was quite busy lately. It seems that bulk transfer fails, initialize goes well. I have only one pro camera at my disposal so I never tested it on a second camera. In this way, I can only guess what is going wrong at your side. If possible, can you create maybe a wireshark dump of the usb traffic while running the program? Even better would be that you could capture the usb traffic when running the seek android app. I did this by running the android app on an android emulator (genymobile) on my linux desktop. If I then started wireshark on my desktop (after forwarding the usb device to genymobile; see howto here: http://scrumsofanarchy.com/interfacing-physical-otg-devices-to-genymotion-android-vm/), I could see the usb traffic. This allowed me to mimic the commands send by the seek app.
Sorry that I cannot help you any further in a more easy way.

maartenvds, 2017-05-18:
By the way, did you try to increase the timeout for bulk transfer? You can do this in SeekDevice.h line 68

closed 2017-05-25 by maartenvds