OpenThermal/libseek-thermal #6: Lots of artifacts in the image

fnoop, 2017-05-20:
I see lots of dots/artifacts in the image.  These are static - the all remain in the same place - no matter where the camera is pointed.  They are not present in the libseek images.  Am I missing a step somewhere to correct bad pixels or something?
[image]

maartenvds, 2017-05-20:
Can you give a bit more context? What camera are you using and are you using the provided samples or code of your own? Is the rest of the image as expected besides the white dots (since you seem to be pointing at a cold background)? library handles bad pixels so normally no need to correct them.

fnoop, 2017-05-21:
Sorry, this is an original seek XR (believe it's now called compact XR?), using the standard 'seek_test' code.  As you can see below the picture is excellent, apart from these artifacts.  They seem to be the same on different invocations and even after it has been turned off for a long time.
[image]

maartenvds, 2017-05-21:
Looks like the dead pixel filter is doing a bad job :D. I created a new branch called 'dead_pixel_test' where I added some code that writes out 3 .png files. Can you compile this code and run it on your camera using the 'seek_test' executable? It will exit automatically once it has written the 3 .png files in the current directory. If you then send me the 3 files in a zip/tar container, I'm able to reproduce what you are seeing.

fnoop, 2017-05-21:
Done, attached, thanks :)
[dead_pixel_frames.zip](https://github.com/maartenvds/libseek-thermal/files/1016767/dead_pixel_frames.zip)

maartenvds, 2017-05-23:
I found the problem, a bit of a naive implementation of me. Currently working on a solution.

maartenvds, 2017-05-23:
Hi, I fixed the dead pixel filter and tested it on your camera images. It looks as expected. The fix is on the development branch. Can you verify the patch? I still have to test it on my camera before merging it to master but I currently don't have it here.

fnoop, 2017-05-23:
Confirmed fixed my dead pixels.  Thanks so much :)

closed 2017-05-25 by maartenvds