OpenThermal/libseek-thermal #7: Include pkgconfig/cmake files?

fnoop, 2017-05-21:
It would be great if install target would create and install pkgconfig .pc files for pkgconfig, and/or .cmake files for cmake, so other code can easily find the installed library/includes.

maartenvds, 2017-05-23:
libseek.pc created and now part of the install target

fnoop, 2017-05-23:
Cflags sets fixed include path, it should honour install PREFIX location:
`Cflags: -std=c++11 -I/usr/local/include/seek`

maartenvds, 2017-05-24:
So true, fix on development branch

fnoop, 2017-05-24:
Libs also missing link and runpaths:
`Libs: -L/srv/maverick/software/libseek-thermal/lib -R/srv/maverick/software/libseek-thermal/lib -lseek`

maartenvds, 2017-05-24:
I did add
```
-L${prefix}/lib
```
to the Libs in the last commit. Is this what you mean? I however did not know about the existense of a -R path. Will this work if I just add it to the Libs? I read somewhere that gcc does not know this option and since it is then printed anyway when you ask pkg-config about the --libs, your linker command might fail if using gcc

fnoop, 2017-05-24:
Hmm, it always used to be the case you just set -Lpath and -Rpath during linking, but seems that's changed now to -Lpath -Wl,-rpath=path --enable-new-dtags.  So:
`Libs: -Wl,-rpath=/srv/maverick/software/libseek-thermal/lib --enable-new-dtags -L/srv/maverick/software/libseek-thermal/lib -lseek`

This gives:
```
$ objdump -x vision_seek |grep -i rpath
  RPATH                /srv/maverick/software/libseek-thermal/lib
```

maartenvds, 2017-05-24:
I would like to add this for you, but when I then do the following after install:
```
g++ test_pro.cpp -o test_pro `pkg-config libseek --libs --cflags`
```
I get:
```
cc1plus: error: unknown pass new-dtags specified in -fenable
``` 
It works when I leave out the '--enable-new-dtags'. Not sure why this is happening...

maartenvds, 2017-05-24:
I got it to work with
```
Libs: -Wl,--enable-new-dtags,-rpath=${prefix}/lib
```
the flag should be prefixed with -Wl. See latest commit
I however don't see the rpath in my executable if I try your objdump command, not sure why that is. Can you confirm if this is working or not?

fnoop, 2017-05-24:
OK reading up on this it's probably better for your library that you don't try to set runpaths in the pkgconfig, but rather rely on ldconfig.  If the coder wants to specifically set the runpath then that's up to them to set as they wish - some will want rpath, some will want runpath.
But I do think it would be good idea to set the link time path:

Libs: -L${prefix} -lseek

maartenvds, 2017-05-24:
Done :)

closed 2017-05-25 by maartenvds