OpenThermal/libseek-thermal #8: Runtime support for different camera types

fnoop, 2017-05-22:
The difference between test and testpro is:
    LibSeek::SeekThermal seek(argc == 2 ? argv[1] : "");
and
    LibSeek::SeekThermalPro seek(argc == 2 ? argv[1] : "");

It would be nice to be able to declare seek something like:
    LibSeek::Seek seek("pro");

So we don't have to write different code for different models.

maartenvds, 2017-05-23:
You have runtime support if you want using polymorphism.
Example with smart pointers:
```
std::shared_ptr<LibSeek::SeekCam> seek;

if (camtype == "pro")
        seek = std::make_shared<LibSeek::SeekThermalPro>();
else
        seek = std::make_shared<LibSeek::SeekThermal>();

if (!seek->open())
...
```

Or just on the stack:

```
LibSeek::SeekCam* seek;
LibSeek::SeekThermalPro seekpro;
LibSeek::SeekThermal seekclassic;

if (camtype == "pro")
        seek = &seekpro;
else
        seek = &seekclassic;

if (!seek->open())
```

I could add a factory function that does this behind the scenes, but personally I think that would not add any real value, especially since the lib currently only supports two camera types.

fnoop, 2017-05-23:
Aaaah!  Thanks so much, I spent ages trying to get both of those approaches to work.  My C++ sucks :)

fnoop, 2017-05-23:
Works perfectly, thanks :)

closed 2017-05-23 by fnoop