OpenThermal/libseek-thermal #13: seek-viewer w/ SEEK Compact Pro image size

jmb365, 2017-08-23:
Thank you for making this software available!

When  try seek-viewer with a SEEK Thermal Compact Pro I do not seem to get the 320x240 image like the seek-Test program does.  What am I doing wrong?

Also what is the syntax for "--output= "?  When I tried it with "--output=/tmp/Seek.avi"  the resulting file is only about 5Kbytes regardless of the duration of the capture and is unreadable by vlc or a media player.  I am using Ubuntu 16.04 (64bit) OS.

Would very much appreciate some help.  Thank you!

jmb365, 2017-08-25:
Ignore my question about the image size.  I wasn't using the --rotate command line option.

However, I am still struggling with the video capture file issue.  I am not a c++ programmer, so any help would be very useful.

maartenvds, 2017-08-25:
Works fine with me, same os as yours. Can you give some more details:
* What version of opencv are you using and how did you installed it
* Compiler output when you compiled libseek-thermal
* The exact full command you use to run the camera

jmb365, 2017-08-25:
Thank you for your quick response!  Here are the details:
=====================================================================
# Dependencies (as shown in GitHub README:
#    libopencv-dev (>= 2.4)
#    libusb-1.0-0-dev
#    libboost-program-options-dev

My steps are listed below ...
=====================================================================
sudo aptitude install libopencv-dev libusb-1.0-0-dev libboost-program-options-dev

sudo aptitude search libopencv-dev
# i   libopencv-dev                                            - development files for opencv                                       
# p   libopencv-dev:i386                                       - development files for opencv 
sudo aptitude install apt-show-versions
apt-show-versions libopencv-dev
# libopencv-dev:amd64/xenial 2.4.9.1+dfsg-1.5ubuntu1 uptodate
# libopencv-dev:i386 not installed

make 
# make -C lib/
# make[1]: Entering directory '/home/userid/Desktop/Junk/Seek/libseek-thermal/lib'
# g++ -I/usr/include/opencv -I/usr/include/libusb-1.0 -Wall -std=c++11 -O2 -fPIC   -c -o SeekThermalPro.o SeekThermalPro.cpp
# g++ -I/usr/include/opencv -I/usr/include/libusb-1.0 -Wall -std=c++11 -O2 -fPIC   -c -o SeekThermal.o SeekThermal.cpp
# g++ -I/usr/include/opencv -I/usr/include/libusb-1.0 -Wall -std=c++11 -O2 -fPIC   -c -o SeekDevice.o SeekDevice.cpp
# g++ -I/usr/include/opencv -I/usr/include/libusb-1.0 -Wall -std=c++11 -O2 -fPIC   -c -o SeekCam.o SeekCam.cpp
# g++ -shared -o libseek.so SeekThermalPro.o SeekThermal.o SeekDevice.o SeekCam.o -lstdc++ /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so -lopencv_calib3d /usr/lib/x86_64-linux-gnu/libopencv_contrib.so -lopencv_contrib /usr/lib/x86_64-linux-gnu/libopencv_core.so -lopencv_core /usr/lib/x86_64-linux-gnu/libopencv_features2d.so -lopencv_features2d /usr/lib/x86_64-linux-gnu/libopencv_flann.so -lopencv_flann /usr/lib/x86_64-linux-gnu/libopencv_gpu.so -lopencv_gpu /usr/lib/x86_64-linux-gnu/libopencv_highgui.so -lopencv_highgui /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so -lopencv_imgproc /usr/lib/x86_64-linux-gnu/libopencv_legacy.so -lopencv_legacy /usr/lib/x86_64-linux-gnu/libopencv_ml.so -lopencv_ml /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so -lopencv_objdetect /usr/lib/x86_64-linux-gnu/libopencv_ocl.so -lopencv_ocl /usr/lib/x86_64-linux-gnu/libopencv_photo.so -lopencv_photo /usr/lib/x86_64-linux-gnu/libopencv_stitching.so -lopencv_stitching /usr/lib/x86_64-linux-gnu/libopencv_superres.so -lopencv_superres /usr/lib/x86_64-linux-gnu/libopencv_ts.so -lopencv_ts /usr/lib/x86_64-linux-gnu/libopencv_video.so -lopencv_video /usr/lib/x86_64-linux-gnu/libopencv_videostab.so -lopencv_videostab -lusb-1.0
# ar rcs libseek.a SeekThermalPro.o SeekThermal.o SeekDevice.o SeekCam.o
# make[1]: Leaving directory '/home/userid/Desktop/Junk/Seek/libseek-thermal/lib'
# make -C bin/
# make[1]: Entering directory '/home/userid/Desktop/Junk/Seek/libseek-thermal/bin'
# g++ -I/usr/include/opencv -I/usr/include/libusb-1.0 -Wall -std=c++11 -O2 -I../lib   -c -o seek_test_pro.o seek_test_pro.cpp
# g++ -lstdc++  seek_test_pro.o ../lib/libseek.a  /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so -lopencv_calib3d /usr/lib/x86_64-linux-gnu/libopencv_contrib.so -lopencv_contrib /usr/lib/x86_64-linux-gnu/libopencv_core.so -lopencv_core /usr/lib/x86_64-linux-gnu/libopencv_features2d.so -lopencv_features2d /usr/lib/x86_64-linux-gnu/libopencv_flann.so -lopencv_flann /usr/lib/x86_64-linux-gnu/libopencv_gpu.so -lopencv_gpu /usr/lib/x86_64-linux-gnu/libopencv_highgui.so -lopencv_highgui /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so -lopencv_imgproc /usr/lib/x86_64-linux-gnu/libopencv_legacy.so -lopencv_legacy /usr/lib/x86_64-linux-gnu/libopencv_ml.so -lopencv_ml /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so -lopencv_objdetect /usr/lib/x86_64-linux-gnu/libopencv_ocl.so -lopencv_ocl /usr/lib/x86_64-linux-gnu/libopencv_photo.so -lopencv_photo /usr/lib/x86_64-linux-gnu/libopencv_stitching.so -lopencv_stitching /usr/lib/x86_64-linux-gnu/libopencv_superres.so -lopencv_superres /usr/lib/x86_64-linux-gnu/libopencv_ts.so -lopencv_ts /usr/lib/x86_64-linux-gnu/libopencv_video.so -lopencv_video /usr/lib/x86_64-linux-gnu/libopencv_videostab.so -lopencv_videostab -lusb-1.0 -lboost_program_options -o seek_test_pro
# g++ -I/usr/include/opencv -I/usr/include/libusb-1.0 -Wall -std=c++11 -O2 -I../lib   -c -o seek_test.o seek_test.cpp
# g++ -lstdc++  seek_test.o ../lib/libseek.a  /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so -lopencv_calib3d /usr/lib/x86_64-linux-gnu/libopencv_contrib.so -lopencv_contrib /usr/lib/x86_64-linux-gnu/libopencv_core.so -lopencv_core /usr/lib/x86_64-linux-gnu/libopencv_features2d.so -lopencv_features2d /usr/lib/x86_64-linux-gnu/libopencv_flann.so -lopencv_flann /usr/lib/x86_64-linux-gnu/libopencv_gpu.so -lopencv_gpu /usr/lib/x86_64-linux-gnu/libopencv_highgui.so -lopencv_highgui /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so -lopencv_imgproc /usr/lib/x86_64-linux-gnu/libopencv_legacy.so -lopencv_legacy /usr/lib/x86_64-linux-gnu/libopencv_ml.so -lopencv_ml /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so -lopencv_objdetect /usr/lib/x86_64-linux-gnu/libopencv_ocl.so -lopencv_ocl /usr/lib/x86_64-linux-gnu/libopencv_photo.so -lopencv_photo /usr/lib/x86_64-linux-gnu/libopencv_stitching.so -lopencv_stitching /usr/lib/x86_64-linux-gnu/libopencv_superres.so -lopencv_superres /usr/lib/x86_64-linux-gnu/libopencv_ts.so -lopencv_ts /usr/lib/x86_64-linux-gnu/libopencv_video.so -lopencv_video /usr/lib/x86_64-linux-gnu/l [... cut ...] x-gnu/libopencv_ts.so -lopencv_ts /usr/lib/x86_64-linux-gnu/libopencv_video.so -lopencv_video /usr/lib/x86_64-linux-gnu/libopencv_videostab.so -lopencv_videostab -lusb-1.0 -lboost_program_options -o seek_create_flat_field
# make[1]: Leaving directory '/home/userid/Desktop/Junk/Seek/libseek-thermal/bin'

uname -a
# Linux 4.4.0-92-generic #115-Ubuntu SMP Thu Aug 10 09:04:33 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

lsb_release -a
# Description:	Ubuntu 16.04.2 LTS
# Release:	16.04
# Codename:	xenial

jmb365, 2017-08-25:
Sorry about the large font!  I do not know what caused it or how to fix it...

jmb365, 2017-08-25:
Test of code embedding:

<This is code>

jmb365, 2017-08-31:
Hello Maartenvds,

Did you get the chance to look at this problem yet?  Is there any other information you need?

Thank you for your assistance.

maartenvds, 2017-09-14:
Hi there, sorry for the late reply. I've been quite busy lately but now I had some time to look into it. I succeeded in reproducing the issue. It works with opencv 3.x but it does not with opencv 2.4. I've pushed a fix on the development branch, can you verify if this solves your problem? It will save video in compressed format instead of raw.
Please note that I'm changing to cmake build instead of GNU make. So on the development branch you will have to build the code using cmake. You can find instructions on the REAMDE page of the development branch to guide you.

jmb365, 2017-09-19:
Hello,
You have fixed it.  It works beautifully... Thank you!
Are you going to merge it into a main code, or will you wait till the next release?

maartenvds, 2017-11-05:
Done ;)

closed 2017-11-05 by maartenvds

krismc, 2017-11-17:
Intuiting from above I'm trying for something from my seek thermal compact with 
seek_viewer --camtype=seek --colormap=2 --rotate=0 --FFC=seek_ffc.png --output=Seek.mp4
which gives [mp4 @ 0x3e08c0] Tag FMP4/0x34504d46 incompatible with output codec id '13'
or
seek_viewer --camtype=seek --colormap=2 --rotate=0 --FFC=seek_ffc.png --output=Seek.png 
which gives [image2 @ 0x7528c0] Could not get frame filename number 2 from pattern 'Seek.png'

Pardon the pun, but am I getting warm?