OpenThermal/libseek-thermal #15: Frame ID never changes from 4

zeroKilo, 2017-09-19:
```
bool SeekCam::grab()
{
    int i;

    for (i=0; i<10; i++) {
        if(!get_frame()) {
            error("Error: frame acquisition failed\n");
            return false;
        }

        if (frame_id() == 3) {
            return true;

        } else if (frame_id() == 1) {
            m_raw_frame.copyTo(m_flat_field_calibration_frame);
        }
    }

    return false;
}
```

if I understand this code right, it waits for the incoming frame data to have a frame id of 3 (3rd 16bit value in frame) but my camera never changes from 4, what would trigger it to change? I made a few edits to the code to have better output:

https://pastebin.com/hEB2wuFV

it fails with:

`/home/tester/code/thermal/2/src/SeekCam.cpp:174:open_cam(): Error: first grab failed`

what can I do? is there a documentation to the frame data structure?

maartenvds, 2017-09-19:
Did you try this from a linux virtual machine or native linux? What camera type did you use?

zeroKilo, 2017-09-19:
I used linux mint in vmware player, and I have PID=0011, "Seek Thermal Compact Pro"

AchrefDash, 2018-02-27:
Hi, i have the same problem here, did you find a solution ?
Ty

maartenvds, 2018-02-27:
Nope, sorry it is known not to work stable via a virtual machine. I'm currently not planning to provide support for VM's.

AchrefDash, 2018-03-05:
Ok Thanks. 
I still have one question, i want to implement an Android App base on your c++ code. Two solutions came to my mind. The first one is to translate your code in java and the second one consists of using NDK to integrate you code as a native code. Wich one do you recommand me? And please tell me if you have any other solution to make your code running on Android devices.

twhitloc, 2018-03-20:
I am facing a similar issue with an Jetson TX2, I can hear the shutter on the camera activate and yet it seems to fail the image grab every time.

> /home/nvidia/libseek-thermal/src/SeekDevice.cpp:149:open_device(): Device Count : 3
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:159:open_device(): vendor: 1d6b  product: 3
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:159:open_device(): vendor: 289d  product: 11
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:57:open(): bConfigurationValue : 1
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x54, 0x0, 0x0, 1)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x4e, 0x0, 0x0, 4)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:   
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x36, 0x0, 0x0, 12)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  
 
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 12)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  0 A 1 2 A 1 J U L F 5 4
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x55, 0x0, 0x0, 2)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x4e, 0x0, 0x0, 64)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  0 x D 1 5 D D 3 8 F 1 2 0 A 
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 2)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 2)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:   
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:   
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  V    
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  f O  = �  ▒ 7 : 0 � � � �  B � � b ) � h � � & � K � � n � / * u ` � � � R < �   � 7 S  � � �  � B 
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
/home/nvidia/libseek-thermal/src/SeekCam.cpp:212:print_usb_data(): Response:  
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x56, 0x0, 0x0, 6)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0xc1, 0x58, 0x0, 0x0, 64)
[... cut ...]
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:125:fetch_frame(): Actual length 13680
/home/nvidia/libseek-thermal/src/SeekCam.cpp:174:open_cam(): Error: first grab failed
failed to open seek cam
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)
/home/nvidia/libseek-thermal/src/SeekDevice.cpp:199:control_transfer(): ctrl_transfer to/from dev(0x41, 0x3c, 0x0, 0x0, 2)