OpenThermal/libseek-thermal #16: How to get the usbcommand

Tonykids, 2017-11-28:


maartenvds, 2017-11-28:
Sniffing USB trafic with wireshark on a host machine while the android app was running on a virtual android machine (on the same host), communicating with the camera. I just mimicked the behavior of the initialization procedure.

Tonykids, 2017-11-28:
I tried vmware workstation and vitrualbox but when i connect the device into the virtual machine , the 
 device do not show in wireshark  on the host.
my host is windows 7,or I need a Linux platform?

maartenvds, 2017-11-28:
I used genymotion android emulator on a native linux machine (ubuntu). Wireshark was running on the native linux. In addition I had to forward the usb device to the emulator so it could access it. You could install genymotion on windows and try the same thing. Never did that myself though.

Tonykids, 2017-11-29:
Succeed !
Thanks a lot !

closed 2017-12-01 by maartenvds