OpenThermal/libseek-thermal #17: Usage

Qo2770, 2017-12-01:
Hello,

I would like to know how I could directly obtain temperature values for pixels. If temperature values are not currently implemented, I would need to access all pixels (like in an array or similar) and run checks. How would I do that? 

Also, is there a general documentation yet, or is it best to just look at the code of the examples and move forward from there?

Thanks,
-Quentin Oschatz

P.S.: I have a Seek Thermal Compact

maartenvds, 2017-12-01:
I'm affraid this is not supported. I might need to add this to the readme which besides that includes everything you need to know about the usage of the driver. Please look at the test_seek en test_seek_pro examples. The API returns the raw 14-bit pixel values which contain the absolute temperature measurements. You can use them to manually calibrate real temperature measurements if you want.

alxpettit, 2020-08-03:
#39 is a duplicate of this issue, but #39 is far more descriptive. Recommend closing.

closed 2020-08-09 by Bostwickenator