OpenThermal/libseek-thermal #19: Cannot work out image format

fnoop, 2017-12-10:
Using seek_viewer with the --output argument, this uses OpenCV VideoWriter (and hence gstreamer) to output the frames through a provided gstreamer pipeline.  A pipeline such as this should work (on a raspberry):
```
OUTPUT="appsrc ! queue ! autovideoconvert ! video/x-raw,format=I420 ! omxh264enc ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=192.168.1.xxx port=5600 sync=false"
```
 with the client pipeline:
```
gst-launch-1.0 -v udpsrc port=5600 buffer-size=0 caps="application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" ! application/x-rtp,encoding-name=H264,payload=96 ! rtph264depay ! queue ! avdec_h264 ! autovideosink sync=false
```
This is pretty standard gstreamer stuff.  However for seek_viewer I cannot get it to work - I just get a green box with the odd flapping pixel.  I've tried explicitly converting the output frame to RGB but no luck.

Any ideas?  This would be a significant improvement as it allows realtime video over wifi such as would be used in search and rescue etc.

gitty8, 2018-09-26:
Any news here? Im interested in stream the image in realtime via wifi to a smartphone near by.

kristofbremen, 2018-10-08:
I would be interested too

Bostwickenator, 2020-09-05:
Does the v4l2 functionality solve for your usecase?

alxpettit, 2020-09-13:
v4l2 solves this. You can stream to v4l2 and then stream from that to wherever via OBS or VLC or whatever else. I could write up a markdown walkthrough if needed! :)

fnoop, 2020-09-21:
Would be great to have v4l2 support documented in the readme.

Bostwickenator, 2020-09-25:
Documented, hope that helps.

closed 2020-09-25 by Bostwickenator

fnoop, 2020-09-28:
Thankyou :)