OpenThermal/libseek-thermal #20: USB soft disconnection problem

francescoDV90, 2018-01-19:
Hello,
I am using your code to connect my seek thermal compact to a linux platform running on a virtual machine (to access the USB I have tu use sudo). When I run the test code everithing is working properly, the problem is when I exit the code typing Ctrl+C.
The code terminates, but when I run again the .exe, I get an error message about a PIPE error. This error does not terminate the code, that will start running.

The critical problem is that  Ctrl+C does not free completely the USB device, somehow in backgrownd there is something keeping alive the communication as explained here: [http://hertaville.com/introduction-to-accessing-the-raspberry-pis-gpio-in-c.html](url) under **Using Signals**

When I have tried to close the virtual machine, I got an error message that it was busy (this will confirm my hypothesis). I forced the close, and at the restart I got damaged the system files of Ubuntu. I have reinstalled everithing (starting from Ubuntu,libreries etc) twice, and the problem will happen again. So I am pretty sure that there is a bug, or a way to safetly remove the USB connection when we press Ctrl+C, without destroing the PC.

Could you please help me?

maartenvds, 2018-02-07:
The lib runs unfortunatly unstable from a virtual machine, I havent even been able to run it on a virtual machine myself due to errors in the communication so I'm affraid I'm not able to help. I advise you to switch to a native implementation. I you however would find a solution to your problem somehow, I would be glad to implement it. Sorry for the inconvenience

francescoDV90, 2018-02-07:
hello,
thank you for the replay..no worry about that.. I am using the raspberry pi
now, and there a native linux is installed...even if I have to use sudo any
case to access the USB.

maartenvds, 2018-02-07:
Ok than I will close the issue

closed 2018-02-07 by maartenvds