OpenThermal/libseek-thermal #25: Compact Pro on RPI 3 B LIBUSB_ERROR_TIMEOUT

Merkensen, 2018-06-02:
Hi Maarten,

I'm trying to get the Seek Compact Pro running on a Raspberry Pi 3 B using maverick 1.1.6 
but the vision_seek service stops after a few seconds.
timeout for bulk transfer is set to 10000. I tried higher values but with no success.

here is the log for vision_seek:

Displaying service log: vision_seek
-- Logs begin at Thu 2016-11-03 17:16:43 GMT. --
Jun 02 11:58:25 maverick-raspberry systemd[1]: Started Maverick - Seek Thermal Vision.
Jun 02 11:58:25 maverick-raspberry vision_seek.sh[9302]: /srv/maverick/software/libseek-thermal/bin/seek_viewer --fps=7 --scale=1.0 --
colormap=2 --rotate=90 --camtype=seekpro --FFC=/srv/maverick/config/vision/seek_ffc.png -o /srv/maverick/data/vision/vision_seek/seek.avi
Jun 02 11:58:28 maverick-raspberry vision_seek.sh[9302]: Video stream created, dimension: 240x320, fps:7
Jun 02 11:58:59 maverick-raspberry vision_seek.sh[9302]: Error: bulk transfer failed:
LIBUSB_ERROR_TIMEOUT
Jun 02 11:58:59 maverick-raspberry vision_seek.sh[9302]: Error: frame acquisition failed

and the lsusb:

Bus 001 Device 008: ID 289d:0011
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Do you know that the Compact Pro is generally running on a RPI 3 B?
I tried it on a RPI Zero W with the same result.
I hope you can help me with this issue.

maartenvds, 2018-06-02:
I know some guys using this lib succesfully on a pi, not sure what versions though. Does it always fail or sometimes? Have you tried using a different usb cable (thin cheap cables tend to fail in my experience).

Merkensen, 2018-06-02:
Sometimes it runs up to a minute, but it always fails. I'm using an OTG USB2.0 A to micro-USB 5-pin adapter on the PI 3 and a full 5-pin micro-USB extension on the PI 0 W. I would like to try some other cables, but I don't really know what specs they should have.

maartenvds, 2018-06-02:
Try to use a thicker non OTG cable. It's a bit trial and error. If it's still not working stable after trying multiple different cables, try increasing the usb timeout set in the driver. Let me know the outcome

Merkensen, 2018-06-03:
Hi,
I tried another adaptor on the RPI 3 B with the same result. I can't find the right cable to purchase so I'll have to wait for the parts to solder my own.
In the meantime I switched the power-supply on the RPI zero to the GPIO so I am able to plug the compact pro directly into the Pi's micro-USB but I am still getting this timeout error.
On a PC with Win10 using the USB-adaptor and ThermalView and directly on a smartphone it worked fine. 
Did you mean the timeout settings in the SeekDevice.h? I changed that from 10000 (10sec) to 100000 (100sec) and to 1000000 (1000sec). Same result.
Anything else worth a try?

maartenvds, 2018-06-04:
Ok, if you are positive that the usb cable is fine, you could try the following: modify the test_seek_pro example by inserting a sleep in the while loop. It could be that the request for a new frame comes too fast causing the camera to miss it. No request, no respons, timeout. Make sure that you only have like max 5fps or so. Try if this runs longer.

Merkensen, 2018-06-04:
How can I modify the test_seek_pro? If I open it with ssh I'm getting wierd symbols.
Sorry, I'm quite new to this.

maartenvds, 2018-06-04:
Modifying the source ofcourse and recompile (examples dir)

AchrefDash, 2018-06-04:
Hi, 
I have the same issue by here. I modified the seek_test_pro.cpp by adding a sleep in the loop. 
The issue persists.

DoctorSnake, 2018-09-16:
Hi, Maarten, hi everyone. 
I have the same issue too, on Raspberry pi 3.
It happens randomly. Today i made the camera work during 5 min without seeing the error. But sometimes the program crashes after just a few seconds.

I will try the increasing the timeout and will let you know.

microhobby, 2020-03-30:
What cable are you using to connect the camera to the PC or Raspberry Pi?

joshdoe, 2020-05-13:
> What cable are you using to connect the camera to the PC or Raspberry Pi?

Would be great to get a list of supported cables/adapters.

Bostwickenator, 2020-05-13:
I don't think the cables play a large part here. This repo definitely shouldn't be in the business of vetting USB cables. Every cable I've used has worked perfectly.

joshdoe, 2020-05-13:
For clarity then, any USB micro B female to USB male A cable will work? OTG is only needed when connecting to a phone. In other words just the four pins need to be connected, vcc, ground, and data +/-.

Bostwickenator, 2020-05-13:
Yes. On the go just shorts the sense line to ground in order to tell the OTG compatible phone or other device to enter host mode instead of it's default slave mode. If the device uses a USB A port it only supports host mode anyway and this doesn't need to be communicated.

Bostwickenator, 2021-01-02:
Has anyone had this issue recently? If not I'll close it out.

closed 2021-05-16 by Bostwickenator