OpenThermal/libseek-thermal #26: flickering video?

pauledd, 2018-06-04:
Hi
I am just a new owner of the CompactPro and stumbled upon this repo.
When starting the example app...
`seek_viewer -F seekpro_ffc.png -s 2 -r 180 -t seekpro`
...I get very flickery video...
https://www.youtube.com/watch?v=PbQ8qiWEhvg&feature=youtu.be
Do you have this too? My cam works without any issues on my android.

maartenvds, 2018-06-04:
Does the 'flicker' go away when you turn the camera around? What you see is auto contrast. The implemented auto contrast is a simple linear version and not really part of the driver. Some points of view can give this effect.

pauledd, 2018-06-05:
No, sometimes seems to become stable, but only for ~3secs, then flickering continues. Turning the cam does not change that.