OpenThermal/libseek-thermal #27: read raw data method?

pauledd, 2018-06-16:
Hi
I am just practising a little bit with QT and libseek-thermal, but  bear with me, I am a quite bloody C++ beginner...
I successfuly created a little app that streams the cam.
```
...
if(!seekf.read(frame0))
    {
        qDebug() << "no more image...";
        return 1;
    }
    cv::normalize(frame0,frame1, 0, 65536, cv::NORM_MINMAX);
    frame1.convertTo(frame1,CV_8U, 0.00390625);
    for(int i=0;i<img->height();i++)
    {
        memcpy(img->scanLine(i), frame1.row(i).data, img->bytesPerLine());
    }
    lbImage->setPixmap(QPixmap::fromImage(*img));
...
```
So far that works quite good. But how can I obtain a raw frame that I can play with it? 
If I understand correctly seekf.read(frame0) reads already 16bit? greyscale preprocessed frames.
Is there a method to read raw data into a MAT/Array? What I want to do is basically to get the raw min/max values of the 320x240 frame with its corresponding coordinates x/y.

maartenvds, 2018-06-16:
Just get rid of the 'normalise' function. frame0 contains the raw image data in 16bit in a Mat container. There is a pull request with suggested conversion to obtain temperature values in case you are interested. It is still experimental ofcourse.

pauledd, 2018-06-16:
Ah yes, I just noticed its already raw in that pull request code, thanks!

closed 2018-06-16 by pauledd