OpenThermal/libseek-thermal #28: cannot close and re-open LibSeek::SeekThermalPro

pauledd, 2018-06-16:
Hi

After I opened the cam with:
```
if(!cam.open())
    {
        mBox->setText("Unable to open SeekThermal device");
        mBox->show();
        return 1;
    }
```
and I close it later with:
`cam.close();`
I cannot open it with the first command again, I get the error
```
Error: SeekDevice already opened
Error: open failed
```
but I can clearly hear that the shutter closed on the cam.close command...
I am doing it wrong?

AchrefDash, 2018-06-16:
Hi, 
To reopen the camera you can use the close function in seekdevice.cpp to close the usb interface. I defined a new function in seekcam. cpp like this and called it from seek_test_viewer.cpp.

In SeekCam.cpp

void closeDevice ()
    {
        m_dev.close();
    }

In seek_viewer.cpp 

cam.closeDevice(); 

this tip worked for me.

pauledd, 2018-06-17:
okay, thanks, that "dirty" fix works.. somehow almost always. I sometimes get a 
`Error: expected first frame to have id 4`
when I try to re-open the cam... so I have to press close again then it mostly works...

PR #61 (2021-01-02): Close seekdevice when the seekcam is closed
This should fix #28. Camera can be closed and opened as needed with this addition.
closed 2021-01-02 by Bostwickenator