OpenThermal/libseek-thermal #29: temp/raw drop issue

pauledd, 2018-06-17:
Sorry this might be a bit offtopic but that seems the only place where to discuss...
I just wrote a little QT application that uses libseek-thermal (without flat generated).

[image]

I then pointed the cam's middle point to an area that was stable ~26°C and 
the app logged the centerpoint values with an average over 7 values to a file.

I plotted the values over time and that's what I got
[image]

That is a drop of about 10°C over 20 minutes. The original SeekThermal App (Android)
does not show me such a drop, it starts just ~1-2°C off and stays quite stable over time.

So the question is what are they doing in their app to compensate for that drop? 
I assume they also get only raw values from the sensor and the rest is in-app-pre-processing...
Can this be corrected with a flat? Or are they using such a curve that I plotted and
compensate their readings against it over time (given that the curve characteristics stays consistent...)?

undera, 2018-07-19:
I made some experiments around this. What is important is that "sensor value" we got decreases when device temperature increases. I have verified that by cooling down camera and then measuring sensor value. So sensor value can't be direct temperature  value in the same way raw pixel data are.

Another thing to notice is that actual pixel values from camera also go down when camera temperature goes up
My hypothesis is that this "sensor value" might be somehow used for compensating ambient temperature of camera, but real formula is yet to be found.

sjtosco, 2019-06-19:
Hi. I'm working too. 
For Compact XR  (not pro) and one point value near of center, an approximated formula is:

celcius = (0.03424 * raw[100,77] + 64) - 273

However, i'm agree with you. I think that sensor_value is like a reference (maybe a thermistor, with exponential curve). The actual value give me a difference with know reference and like this i can measure temp.
It's a theory. 
Best regards!

Bostwickenator, 2021-01-02:
Marking as duplicate of #39 since the two are so closely coupled