OpenThermal/libseek-thermal #30: Device Temp Sensor (Req.)

pauledd, 2018-06-21:
Hi
I know from joe-c and his windows thermovison software that there is somewhere a temperature sensor in the Seek Compact (Pro) devices. I asked him where the value can be read out and the referred to his dll source code. His uint16 "DevTemp" variable gets read out from two bytes from his raw stream data array data[11] and data[12] (for the Compact Pro)...

The corresponding part can be found in "ClassThermalFrame.cs"(*):
```
 internal ThermalFrame(byte[] data, int Width, int Height) {

...
            if (_isSeekPro) {
                StatusByte = data[4];
                DevTemp = (ushort)(data[11] << 8 | data[10]);
                FrameCnt = (ushort)(data[9] << 8 | data[8]);
            }
            else {
                StatusByte = data[20];
                DevTemp = (ushort)(data[3] << 8 | data[2]);
                FrameCnt = (ushort)(data[81] << 8 | data[80]);
            }
...
```

I tried to look through libseek-thermal to understand where I could tap the temperature value but
I dont get behind the code or dont know what part of "m_raw_data" contains the value.

Any ideas anyone?

(*)http://joe-c.de/pages/posts/version_1.8.0.0_tcamdll_207.php

AchrefDash, 2018-06-27:
Hi, 

Did you find any solution for this?

pauledd, 2018-06-27:
Not yet, but I am still on it. I currently try to talk to my cam via python to better understand all the stuff, I am to dumb for that C++ code :-)

pauledd, 2018-07-01:
Ok, that was quite easy, just add this code to the SeekThermalPro.cpp

```
int SeekThermalPro::device_temp()
{
    return m_raw_data[5];
}
```
and this in public SeekThermalPro.h

`virtual int device_temp();`

Then you can simply get the raw integer temperature value from the camera, but don't be surprised, the value goes down as the device gets hotter... at 24° room temperature mine starts at about ~6400.

The same should work with the non-Pro cams but then you have to try m_raw_data[11], or 12, i am not shure;

AchrefDash, 2018-07-02:
Thank you for your asnwer. I also got values around 6400

AchrefDash, 2018-07-02:
Thank you for your asnwer. 

I also got values around 6400 and it remain stable over time (it dosen't go down even when the device gets hotter). Did you make some additional transformation on this value?

pauledd, 2018-07-02:
no, just reading it... Mine does always go down, but I doing some more tests with different environmental temperatures. Here is some data that I just took from a 24.5°C wall at 23°C room temp. over 45min:
[image]

pauledd, 2018-07-02:
Do you have a Compact or Pro?

AchrefDash, 2018-07-02:
I have the Pro one. 

I've got roughly the sames curve. But i'm still curious to know if your method worked out as well with different environmental temperatures and if matchs with real world temperatures.

pauledd, 2018-07-02:
I've no "method" so far ;) just reading raw values... But with the devTemp we have something we can maybe correlate to and it seems to match quite good the pixel values drop over time. Next thing would be to test how the curve looks in extrem envirom. situations. I think about putting the cam into my fridge (~8°C) and record the values again...

ccskoba, 2018-07-09:
Hi guys. DevTemp is very interesting. I also tried to record DevTemp and room temperature every 10 minutes with my seek compact (not pro).

[image]

I believe that DevTemp represents the sensor temperature. And by using this you will surely get good accuracy.

pauledd, 2018-07-11:
I think I would close this since we now know how to get the dev temp, its not on me to decide whether to implement only the  "device_temp()" function in the code or an complete algorithm to correct the display temperatur values... I think this more more up to the implementer who uses this lib and writes his own application.. But ideas can be put in here:  https://github.com/maartenvds/libseek-thermal/issues/29

closed 2018-07-11 by pauledd

undera, 2018-07-19:
THe method of getting dev temp value, as offered by @pauledd  is incorrect, IMO.
Applying what original `ClassThermalFrame.cs` has, gives completely different values of dev temp...

pauledd, 2018-07-19:
Why? What values do you get? I get the same values with joe-c's thermovision app under windows...

undera, 2018-07-19:
If I copy exactly what in joe's app, I get values that are cycle from quite low to quite high:

[image]

One more thing I noticed with the way of simply getting `m_raw_data[5];` is that value "resets" each time I restart cam. So it's definitely not temperature. And I wonder how `m_raw_data[5];` correspond to  `(ushort)(data[11] << 8 | data[10]);` from C#, how did you conclude that?

pauledd, 2018-07-19:
my understanding of
`(ushort)(data[11] << 8 | data[10]);`
is that joe-c data array is ushort little-endian, so you need two byte to form a 16bit value and this is done with this bit-shift <<. The data array in libseek-thermal is already 16bit, so you need also to half the data arrays address if you want to access the corresponding values in a 16bit array, and this is data(5) (16bit) or (10/11) 8bit. You cant simply copy joe-c's code because the array format differs and thus my guess you get that weird values. Then you doing a bitshift on 16bit values...

undera, 2018-07-19:
I have brute-force printed first 20 items if `m_raw_data` and indeed only item 5 behaves as potential sensor. But I fail to understand the meaning of it, because it goes down with the temperature...

pauledd, 2018-07-19:
Maybe its a thermistor, they tend to decrease values as temperature rises...