OpenThermal/libseek-thermal #31: Failed to open seek cam

Aifuding, 2018-07-09:
Elite,can't  open the cam
./seek_test_pro -> Error: first grab failed    failed to open seek cam
help,help..

pauledd, 2018-07-09:
Do you see the cam id via:
`lsusb`
?

Aifuding, 2018-07-09:
yes,  lsusb
Bus 001 Device 002: ID 289d:0011

Aifuding, 2018-07-09:
running on VMware

pauledd, 2018-07-09:
you could try to run seek_test_pro as root... If that works than you need to set up the udev rules...

Aifuding, 2018-07-09:
$ sudo ./seek_test_pro 
Error: first grab failed
failed to open seek cam
Connection method : seek thermal->usb->VMware->lsusb, What can be affected?

pauledd, 2018-07-09:
I have no idea, I run my Seek Pro via qemu without problems on emulated windows... Cant help with VMware

Aifuding, 2018-07-09:
Thank you very much for your kind attention personal response

tejesh002, 2020-04-29:
Error: libusb init failed: LIBUSB_ERROR_ACCESS

closed 2020-09-05 by Bostwickenator