OpenThermal/libseek-thermal #36: Seek Thermal Compact Pro  corrupter frames on Jatson TX2

francescoDV90, 2019-03-14:
I am trying to use the libseek-thermal library for using my Thermal Compact Pro with NVIDIA Jetson TX2. I have correctly used the library on PC and Raspberry PI with no problems. I have installed the library on Jatson TX2 with the last Jetpack and the frames result to be corrupted. There are three horizontal bands in the lower part of the image
as shown in the attached file.

I have tried to install opencv by hand again but without any result.. I can't figure out which could be the cause. It seems that the buffer data is corrupted.

[image]

yacc, 2019-04-05:
I'm experiencing the same issue on Jetson TX2.

francescoDV90, 2019-04-05:
Hello,
Ok perfect. . I suppose it is a driver problem of the jatson. ..
Do you have any possible idea on how to approcch the problem?

yacc, 2019-04-05:
I don't think it's a driver problem.
I got another thermal camera working on the Jetson using libusb1.0.0 and I've seen people report the same issue on other platforms (windows).
My guess, it has to to do with how raw frames are encoded.
I'll post info if I find out more

swapnilsayansaha, 2020-06-16:
Got anything?

swapnilsayansaha, 2020-06-17:
I noticed the same thing. Had to ditch the nano for a raspberry pi. It's a problem with the nano definitely.

francescoDV90, 2020-06-17:
Good morning,
Yes i solved the provlem. It is related to the libseek-thermal library and
not to the sensor or the board..

I god the SDK from Seek thermal..but it is reqired an agreement and a NDA
you have to sign with them

swapnilsayansaha, 2020-06-17:
^ Ah then I am better off with the raspberry pi.. libseek thermal works plug n play on rpi..

zeeblezooble, 2020-07-15:
I had the same issue with both Jetson TX1 and TX2. The problem was with the line 120 in SeekDevice.cpp.
`res = libusb_bulk_transfer(m_handle, 0x81, &buf[done], todo, &actual_length, m_timeout);`
To make this work I changed the number of bytes to receive into the data buffer to the maximum number of bytes that camera can send (as debug information showed):
`res = libusb_bulk_transfer(m_handle, 0x81, &buf[done], 13680, &actual_length, m_timeout);`
Unfortunately it breaks compatibility with other camera models.

rsend, 2020-08-13:
@francescoDV90 if you have the SDK from Seek, can you comment on shutter control? This is something that has been eluding the 3rd-party software for a while last I checked.  Sorry for the OT post - plz feel free to remove.

Bostwickenator, 2020-08-13:
> @francescoDV90 if you have the SDK from Seek, can you comment on shutter control? This is something that has been eluding the 3rd-party software for a while last I checked. Sorry for the OT post - plz feel free to remove.

They cannot they signed an NDA, additionally we wouldn't be able to take that information if it was obtained via a license violation.

rsend, 2020-08-13:
Fair enough - just wanted to know if it was exposed even IN their SDK. That way I / we could get a license to do it, if desired. No malpractise intended.

Bostwickenator, 2020-09-05:
Hi @zeeblezooble do your remember why you said this broke compatibility?
> `res = libusb_bulk_transfer(m_handle, 0x81, &buf[done], 13680, &actual_length, m_timeout);`
> Unfortunately it breaks compatibility with other camera models.

zeeblezooble, 2020-09-06:
> Hi @zeeblezooble do your remember why you said this broke compatibility?
> 
> > `res = libusb_bulk_transfer(m_handle, 0x81, &buf[done], 13680, &actual_length, m_timeout);`
> > Unfortunately it breaks compatibility with other camera models.

Hello, back then this was just my guess. I wasn't sure that size 13680 will work with other cameras, and I had no way to test this.

Bostwickenator, 2020-09-06:
Perfect thanks I just wanted to see if you confirmed that incompatibility or just expected it. I destroyed my regular Seek Compact so I can't test it at this moment. I do know it returns slightly more bytes per request `16224` but requesting less than this shouldn't result in any incompatibility.

zeeblezooble, 2020-09-06:
I've tried requesting less than 13680 bytes (regular packet size) with Seek Thermal Compact Pro. If I remember correctly, that resulted in error

Bostwickenator, 2020-09-06:
Interesting the Jetson and Pi must have stricter USB stacks. I'm seeing no issue with arbitrary sizes a couple of desktop platforms. I'll see if I can validate the changes in #52 and get this mainlined.

closed 2020-09-11 by alxpettit