OpenThermal/libseek-thermal #37: frames missing

ptenax, 2019-07-08:
Hi Maarten,

Thanks very much for this library, I have managed to install and run a SEEK Pro on a Raspberry Pi 3B.  I am having an issue with missing frames.  I use `timeout` to automatically stop a recording and save a file after a set number of seconds, e.g. 20 seconds.  But I always get less than the total duration and total frames expected (as viewed in the Windows software MPC-HC; also VLC).  For example:

    #This gives me 12 seconds of playback and 119 frames
    timeout 20 seek_viewer --fps=10 --camtype=seekpro --colormap=1 --rotate=90 --scale=1 --output=/home/pi/test10fps.avi  

    #This gives me 5 seconds of playback and 124 frames
    timeout 20 seek_viewer --fps=25 --camtype=seekpro --colormap=1 --rotate=90 --scale=1 --output=/home/pi/test25fps.avi

In both cases I used a stopwatch to time the actual recording time, which was indeed 20 seconds.  The internal refresh click on the camera happened several times in each recording.  I am planning to record for sequential 15 minute blocks in my intended application.  I would really like a minimum of 25fps if I can get it.

Advice would be very much appreciated, Kyle

maartenvds, 2019-07-09:
Hi, you will never get 25 fps. These thermal cameras are designed for max 9fps. In practice, you can get a little more, but always less than 20 fps. Also the higher you go, the more instable the framerate will become. My advice, record below 9 fps.

ptenax, 2019-07-09:
Thanks so much Maarten, and again for your great library.

Regards,

Kyle

closed 2020-09-05 by Bostwickenator