OpenThermal/libseek-thermal #41: Can I use this code for Seek shot pro?

darlcode, 2020-04-30:
Hi :) 

I have some questions about this code!
I know this code for seek compact series.
But I wanna know there is a way to connect seek shot pro and stream in real time with this code..!

Thank you for reading :)

Bostwickenator, 2020-05-01:
Check the examples folder to show how to view the video.

ash2703, 2020-06-07:
It is still not clear, can the code be used for seek shot pro

Bostwickenator, 2020-06-07:
This implements a USB interface to for the Seek **Compact** devices. It is not compatible with the Seek Shot series.

ash2703, 2020-06-07:
> This implements a USB interface to for the Seek **Compact** devices. It is not compatible with the Seek Shot series.

Is there any API which supports seek shot pro series

Bostwickenator, 2020-08-09:
Not that I am aware of sorry.

closed 2020-08-09 by Bostwickenator