OpenThermal/libseek-thermal #44: Corrupted frames

swapnilsayansaha, 2020-06-16:
I have connected the seek thermal camera to a Raspberry Pi like board running on Ubuntu. I followed all the instructions. However, I am getting corrupted frames like this:

[image]

[image]

Board connection:

[image]

Anyone knows what's going wrong? I dont see any error in the terminal.

zeeblezooble, 2020-07-15:
Hello, I had a similar issue, and posted solution [here](https://github.com/maartenvds/libseek-thermal/issues/36#issuecomment-658680474).

alxpettit, 2020-09-13:
What board is that? It looks like a jetson nano.

Bostwickenator, 2021-01-02:
Duplicate of #36

closed 2021-01-02 by Bostwickenator