OpenThermal/libseek-thermal #55: Can not stream the thermal video to v4l2 device ?

ToanVNET, 2020-10-15:
I executed this command: 
`./seek_viewer --camtype=seekpro --colormap=11 --mode=v4l2 --output=/dev/video0`
And I got this error: 
```
Error: control transfer failed: LIBUSB_ERROR_PIPE
Error: LIBUSB_ERROR_TIMEOUT
VIDIOC_G_FMT error: Invalid argument
```
How can I fix this ?
/dev/video0 is my webcam. 
/dev/video2 is not listed on /dev/
I also tried with them but it's not working. I understand that after execute this command, and assume that it's working good. So I can use opencv with python to view the stream video from thermal camera. Did I understand right ?

Bostwickenator, 2020-10-15:
First you need to have the v4l2 loopback module installed. This should generate a virtual device which what you should specify in the seek_viewer arguments. It sounds like you are trying to stream to an existing physical webcam which isn't the way v4l2 works. When you have v4l2loopback working you will be able to receive that stream in opencv yes.

ToanVNET, 2020-10-16:
Thank you so much for your support. It's working like a charm now. Have a nice weekend.

closed 2020-10-16 by ToanVNET