OpenThermal/libseek-thermal #57: win10 64 atom vs2015 gui error,something missing not very undestood, how to slove it

714902777, 2020-12-28:
-- Selecting Windows SDK version  to target Windows 10.0.18363.
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Build type:
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- OpenCV ARCH: x64
-- OpenCV RUNTIME: vc14
-- OpenCV STATIC: OFF
-- Found OpenCV 3.4.10 in C:/opencv/build/x64/vc14/lib
-- You might need to add C:\opencv\build\x64\vc14\bin to your PATH to be able to run your applications.

-----------------------------------------------------------------------------
-- The following external packages were located on your system.
-- This installation will have the extra features provided by these packages.
-----------------------------------------------------------------------------
   * OpenCV - Required to handle image processing

-----------------------------------------------------------------------------
-- The following REQUIRED packages could NOT be located on your system.
-- You must install these packages before continuing.
-----------------------------------------------------------------------------
   * libusb  <https://libusb.info/>
     Required to communicate via USB

-----------------------------------------------------------------------------

CMake Error at cmake/MacroLogFeature.cmake:141 (MESSAGE):
  Exiting: Missing Requirements
Call Stack (most recent call first):
  CMakeLists.txt:43 (macro_display_feature_log)

-- Configuring incomplete, errors occurred!
See also "C:/libseek-thermal-master/build/CMakeFiles/CMakeOutput.log".

714902777, 2020-12-28:
what   does this mean and what should I get to do, it is a  64G pad, a small tablet

Bostwickenator, 2020-12-28:
Hey I just ran into this as well. I'm going to assume you followed the readme. The issue is that the files in the libusb package have moved around in recent versions. 

1. Open the directory you extracted libusb into.
2. Copy the files in VS2015 (or the appropriate version for your platform) up out of the VS2015 directory into it's parent. 
3. Regenerate the CMake cache in VS.

I'll fix the Cmake scripts so they can locate these files or add a warning.

714902777, 2020-12-28:
Which files into whose parent, can you show it directly. And in VS there is a lot of error. I am a new starter may not understand your idea.

714902777, 2020-12-28:
My C:\ is almost full, only 3GB left extra. Why not publish a released. What should i do.

Bostwickenator, 2020-12-29:
copy `libusb-1.0.24/VS2015/*.*` to `libusb-1.0.24/`

See to see what it is searching for look at https://github.com/OpenThermal/libseek-thermal/blob/master/cmake/FindLibUSB.cmake#L82

closed 2020-12-30 by Bostwickenator