OpenThermal/libseek-thermal #58: seven errors and  1630 alerts, this is another code, release in vs error. the last problem is the driver file , i install the seek scan from the website of seek thermal, how did i slove it

714902777, 2020-12-29:
1>------ 已启动生成: 项目: ZERO_CHECK, 配置: Debug x64 ------
1>  Checking Build System
2>------ 已启动生成: 项目: seek_static, 配置: Debug x64 ------
3>------ 已启动生成: 项目: seek, 配置: Debug x64 ------
2>  Building Custom Rule C:/libseek-thermal-temp-info_compact/src/CMakeLists.txt
2>cl : 命令行 error D8021: 无效的数值参数“/Wextra”
4>------ 已启动生成: 项目: seek_create_flat_field, 配置: Debug x64 ------
5>------ 已启动生成: 项目: seek_raw_viewer, 配置: Debug x64 ------
6>------ 已启动生成: 项目: seek_test, 配置: Debug x64 ------
4>  Building Custom Rule C:/libseek-thermal-temp-info_compact/examples/CMakeLists.txt
4>cl : 命令行 error D8021: 无效的数值参数“/Wextra”
7>------ 已启动生成: 项目: seek_test_pro, 配置: Debug x64 ------
3>  Building Custom Rule C:/libseek-thermal-temp-info_compact/src/CMakeLists.txt
7>  Building Custom Rule C:/libseek-thermal-temp-info_compact/examples/CMakeLists.txt
3>cl : 命令行 error D8021: 无效的数值参数“/Wextra”
8>------ 已启动生成: 项目: seek_viewer, 配置: Debug x64 ------
7>cl : 命令行 error D8021: 无效的数值参数“/Wextra”
5>  Building Custom Rule C:/libseek-thermal-temp-info_compact/examples/CMakeLists.txt
8>  Building Custom Rule C:/libseek-thermal-temp-info_compact/examples/CMakeLists.txt
5>cl : 命令行 error D8021: 无效的数值参数“/Wextra”
8>cl : 命令行 error D8021: 无效的数值参数“/Wextra”
6>  Building Custom Rule C:/libseek-thermal-temp-info_compact/examples/CMakeLists.txt
6>cl : 命令行 error D8021: 无效的数值参数“/Wextra”
9>------ 已启动生成: 项目: ALL_BUILD, 配置: Debug x64 ------
9>  Building Custom Rule C:/libseek-thermal-temp-info_compact/CMakeLists.txt
========== 生成: 成功 2 个，失败 7 个，最新 0 个，跳过 0 个 ==========

714902777, 2020-12-30:
what does this mean, and what should i do  next.

Bostwickenator, 2020-12-30:
Howdy again I closed your other issue as it appears you have gotten past that error. Also I moved your problem description into your first comment.
- Can I ask which version of VS you are using? (I've only verified VS2019 so far)
- You cannot use this with the Seek Scan. This application only works for the Seek Compact, Compact XR, and Compact Pro. You do not need to install drivers from Seek. You should install the WinUSB driver using Zadig see [the end of the windows section in the readme](https://github.com/OpenThermal/libseek-thermal/blob/master/README.md#windows)

714902777, 2020-12-30:
i use the seek thermal pro. So it is not suitable right.

714902777, 2020-12-30:
After i install the seek scan, i need to unstall the driver. I use vs2015

Bostwickenator, 2020-12-31:
I want to help you out and there is a clearly a bit of a timezone issue here. I suspect you are having issues because CMake support in VS2015 was not very good. Here is a build of seek_viewer for you to try https://drive.google.com/drive/folders/18wrN-1DpYtGcJLl46mvSPGuW_KUjkCEQ it's a x64 debug build, built against libusb-1.0.24 and OpenCV4.5.1 and I've included those DLLs in that folder. That should let you figure out if your hardware is compatible. In the future I suggest you try compiling in VS2019.

714902777, 2020-12-31:
Lacking of VCruntime140_1d.dll and vcruntime140d.dll after my click. Hope could be updated.

714902777, 2020-12-31:
My pc is lacking of vcruntime140_1d.dll,and vcruntime140d.dll is contained in pc, i download the first one from internet, with a flash black windows.

Bostwickenator, 2020-12-31:
Visual studio Visual C++ should have installed those DLLs 🤔. Please run `seek_viewer.exe --help` from the command line to see what messages are emitted from the application. If nothing is displayed you may be missing another DLL. If this displays the help then run `seek_viewer.exe` and see if it complains about not being able to find the camera.

714902777, 2021-01-01:
i build your recently updated code and build, LIBUSB_INCLUDE_DIRS is that the location of libusb.dll? write into the environment path. i unstall the driver and resinstall form zadig. but still not work.
Selecting Windows SDK version  to target Windows 10.0.18363.
The C compiler identification is MSVC 19.0.24215.1
The CXX compiler identification is MSVC 19.0.24215.1
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working C compiler: C:/Program Files (x86)/VC/bin/x86_amd64/cl.exe - skipped
Detecting C compile features
Detecting C compile features - done
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Check for working CXX compiler: C:/Program Files (x86)/VC/bin/x86_amd64/cl.exe - skipped
Detecting CXX compile features
Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

Build type: 
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
LIBUSB_INCLUDE_DIRS Not Found
OpenCV ARCH: x64
OpenCV RUNTIME: vc14
OpenCV STATIC: OFF
Found OpenCV: C:/opencv/build (found version "3.4.10") 
Found OpenCV 3.4.10 in C:/opencv/build/x64/vc14/lib
You might need to add C:\opencv\build\x64\vc14\bin to your PATH to be able to run your applications.

-----------------------------------------------------------------------------
-- The following external packages were located on your system.
-- This installation will have the extra features provided by these packages.
-----------------------------------------------------------------------------
   * OpenCV - Required to handle image processing

-----------------------------------------------------------------------------
-- The following REQUIRED packages could NOT be located on your system.
-- You must install these packages before continuing.
-----------------------------------------------------------------------------
   * libusb  <https://libusb.info/>
     Required to communicate via USB

-----------------------------------------------------------------------------

CMake Error at cmake/MacroLogFeature.cmake:141 (MESSAGE):
  Exiting: Missing Requirements
Call Stack (most recent call first):
  CMakeLists.txt:42 (macro_display_feature_log)

Configuring incomplete, errors occurred!
See also "C:/libseek-thermal-master/build/CMakeFiles/CMakeOutput.log".

714902777, 2021-01-01:
Seek Thermal Viewer

  OPTIONS:

      -h, --help                        Display this help menu
      -m[mode], --mode=[mode]           The mode to use - v4l2, window, file
      -o[output], --output=[output]     Name of the file or video device to
                                        write to
      -F[FFC], --FFC=[FFC]              Additional Flat Field calibration -
                                        provide ffc file
      -f[fps], --fps=[fps]              Video Output FPS - Kludge factor
      -s[scaling], --scale=[scaling]    Output Scaling - multiple of original
                                        image
      -c[colormap],
      --colormap=[colormap]             Color Map - number between 0 and 21
                                        (see: cv::ColormapTypes for maps
                                        available in your version of OpenCV)
      -r[rotate], --rotate=[rotate]     Rotation - 0, 90, 180 or 270 (default)
                                        degrees
      -t[camtype], --camtype=[camtype]  Seek Thermal Camera Model - seek or
                                        seekpro

Bostwickenator, 2021-01-02:
That last comment seems to imply that you have the executable working? 

LIBUSB_INCLUDE_DIRS is the directory which includes the header files. Example: `C:\libusb-1.0.24\include\libusb-1.0` these are defined in FindLibUSB.cmake 

```
# The following standard variables get defined:
#  LIBUSB_FOUND:            true if LibUSB was found
#  LIBUSB_HEADER_FILE:      the location of the C header file
#  LIBUSB_INCLUDE_DIRS:     the directorys that contain headers
#  LIBUSB_LIBRARIES:        the library files
#  LIBUSB_VERSION           the detected libusb version
#  LIBUSB_HAVE_GET_VERSION  True if libusb has libusb_get_version()
```

714902777, 2021-01-05:
I am going to reboot my device, and reinstall the vs2019, good luck

Bostwickenator, 2021-01-17:
@714902777 success?

closed 2021-05-16 by Bostwickenator