OpenThermal/libseek-thermal #63: No able to compile cmake reports not find OpenCV even being installed

afgranero, 2021-01-15:
I followed the instructions and installed the dependencies:

```
sudo apt-get install libusb-1.0-0-dev
sudo apt-get install libopencv-dev

```
But cmake reports: 

> -----------------------------------------------------------------------------
> -- The following REQUIRED packages could NOT be located on your system.
> -- You must install these packages before continuing.
> -----------------------------------------------------------------------------
>    * OpenCV  <https://opencv.org/>
>      Required to handle image processing
> 
> -----------------------------------------------------------------------------

I added those lines to CMakelists.txt:

find_package(OpenCV)

message(STATUS "OpenCV_INCLUDE_DIRS = ${OpenCV_INCLUDE_DIRS}")
message(STATUS "OpenCV_LIBS = ${OpenCV_LIBS}")

getting:

afgranero, 2021-01-15:
I also tried to rewrite the lib in C without using OpenCV for a simples lib.
But my call to libusb_bulk_transfer gives me an LIBUSB_ERROR_IO after connectiong to the Seek CompactPro.

afgranero, 2021-01-16:
I tryed to compile OpenCV from source.
I followed instructions here: http://techawarey.com/programming/install-opencv-c-c-in-ubuntu-18-04-lts-step-by-step-guide/.
I was sucessfull in compiling and installing.
I was able to compile a C++ test program using OpenCV using headers from /usr/include/opencv.

But I am still not able to compile this project, reciving the error of of libopencv-dev not installed.

Bostwickenator, 2021-01-16:
This is interesting because as I'm sure you can see the library does not do anything out of the ordinary to direct CMAKE how to locate OpenCV [we just call](https://github.com/OpenThermal/libseek-thermal/blob/master/CMakeLists.txt#L32) the `find_package` macro.  On most platforms the package will be automatically located. I wonder if something is going wrong with the logging macro. Can you try removing the `TRUE` from `macro_log_feature (OPENCV_FOUND "OpenCV" "Required to handle image processing" "https://opencv.org/" TRUE)` and seeing if the build proceeds.

afgranero, 2021-01-16:
After compiling OpenCV from source I attempted to use cmake again and it worked.
I was able to cmpile the examples.
But I still don't understandt why installling libopencv-dev didn't work

afgranero, 2021-01-16:
> This is interesting because as I'm sure you can see the library does not do anything out of the ordinary to direct CMAKE how to locate OpenCV [we just [...]

I will try.

closed 2021-01-16 by afgranero

duydangg, 2021-07-07:
have anyone solved this problem in win10 
i've downloaded opencv from https://opencv.org/releases/ then install 
i've copied OpenCVConfig.cmake to cmake folder  
which step that i am missing or i'm wrong 
please correct me