OpenThermal/libseek-thermal #64: libseek-thermal shows a camera sensor defect, but official Android app doesn't.

bum112, 2021-03-07:
I have Seek Thermal CompactPRO. I have tested the camera with libseek-thermal and official Android app. The image returned from libseek-thermal looks like camera itself has sensor defect in the bottom, however, Android app doesn't show any defect. I inspected the problem for a very long time in different conditions. I don't think, that it uses very clever AI image repair, what Android app is doing. I have attached the video comparison.

https://user-images.githubusercontent.com/80056442/110251947-bd6ab200-7f8b-11eb-83e2-b549f84f72b4.mp4

What is going on?

Bostwickenator, 2021-03-07:
Very interesting I haven't seen that before on either of my devices. You could attempt to skip the dead pixel filtering and see what happens. See https://github.com/OpenThermal/libseek-thermal/blob/master/src/SeekCam.cpp#L272
just change the body of the function to be `src.copyTo(dst);`. If that changes the result you could either leave it out for your camera or adjust the thresholding functions https://github.com/OpenThermal/libseek-thermal/blob/master/src/SeekCam.cpp#L243

bum112, 2021-03-07:
Holly shit! It actually worked. Thank you! I used the first suggestion.
Could you please add this option in the code, so other people can also use it. I am very big noob.

Just now I have one dead (stuck on white) pixel, but I guess it would be possible to remove that by using your second suggestion.

closed 2021-03-17 by Bostwickenator