OpenThermal/libseek-thermal #71: CompactPRO with USB-C connector produces a strange pattern

samontab, 2022-05-08:
Hi,

Thanks for this excellent library.

I was testing this lib with a CompactPRO camera with a USB-C connector. 

Everything seems to work fine, the USB code is the same as the CompactPRO with a micro USB connector(0011), but there's a constant pattern of wrongly read pixels in the bottom left part of the frame as you can see here:

[image]

The image itself is correctly sized at 240x320, and most of the frame is correctly read with a fast frame rate, but there seems to be something wrong in the actual values of the pixels in the lower left side for some reason. The camera works perfectly fine on the official app on the phone by the way.

Any ideas on how I can solve this would be great, thanks!

MalteBrunn, 2022-06-08:
Hi,
I have the same/similar issue with a regular Compact (non Pro, Micro-USB, pID 0010). 
I'd be happy to provide additional data to fix this.

Bostwickenator, 2022-06-08:
I believe this is the dead pixel replacement. It's decided that all those
pixels need to be replaced. Someone else previously had this issue see #64.
Can you try the fix noted there. If that fixes it I'll add instructions in
the readme and a compilation flag.

samontab, 2022-06-09:
I tried the fix, and it works.

[image]

As reported in #64 now there are some white pixels, but those could probably be removed either by the dead pixel thresholding or simply by masking them out.

Thanks!

MalteBrunn, 2022-06-09:
Awesome!
This helped a lot. I first tried replacing the function body of apply_dead_pixels_filter as described in #64. That gave me an image with some dead pixels obviously.

Afterwards, I changed the threshold in create_dead_pixel_list (https://github.com/OpenThermal/libseek-thermal/blob/c9e09dcde572194c6186f6b896bbcb2c8e29b42b/src/SeekCam.cpp#L240) to
    `const double threshold = hist_max_value.y - (max_value - hist_max_value.y*0.9);`
and together with the original apply_dead_pixels_filter function this works perfectly now.
Thanks!

closed 2023-03-06 by samontab