OpenThermal/libseek-thermal #77: How do I connect my Compact Pro to the pi

couchpotatochip21, 2023-03-23:
I want to use my compact pro with a pi zero. Do I just connect it via usb c? If so is there anything I need to know to prevent damage to my thermal camera? It was gifted to me and I treasure it. I am making a Thermal Hud for a cosplay helmet.

Bostwickenator, 2023-03-23:
Yes connect via USB-C. You don't need to do anything special to protect it.

couchpotatochip21, 2023-03-24:
> Yes connect via USB-C. You don't need to do anything special to protect it.

OK, thank you.

If I am using a pi zero how do I connect it if it only has micro USB.

Bostwickenator, 2023-03-24:
You can Google "usb micro to usb c adapter rpi zero" and buy an adapter
which fits what you are trying to do.

couchpotatochip21, 2023-03-24:
thank you. It is to my understanding that the compact pro and other cameras from seek click and freeze every second or so. Does the rpi help to negate this in any way.

Bostwickenator, 2023-03-24:
No, it does not.