OpenThermal/libseek-thermal #78: Using two seek compacts at once.

randyleram, 2023-09-13:
I’ve got one seek camera working, it’s brilliant. Thanks for the hard work!

Is there a way to run two instances into two select v4l2 loop back devices? I’m not sure where to begin with it.

I currently have a service on boot to start seek viewer streaming to one v4l2 loop back device.

Just unsure how to tell libseek which usb port to use etc.

Hope this made sense.

Bostwickenator, 2023-10-11:
This isn't supported right now unfortunately. It should be possible to add. This https://github.com/OpenThermal/libseek-thermal/blob/master/src/SeekDevice.cpp#L154 is where we walk the devices.
Does running your app a second time fail with "Error: failed to claim interface"?

randyleram, 2023-10-11:
Ok thanks I’ll have a look. Throws a Libusb error trying to run a second time, device busy. Along those lines.