OpenThermal/libseek-thermal #81: control transfer failed: LIBUSB_ERROR_PIPE

blackkarlsson, 2024-09-21:
hey community!

trying to use libseek-thermal for Nano 300 (289d:0011 Seek Thermal, Inc. Nano300) and get the error in issue subj on init cam

is this a bug or Nano 300 isn't supported yet? 

many thanks in advance for helping with the issue!

Bostwickenator, 2024-09-22:
Not supported yet, I haven't looked into it.

blackkarlsson, 2024-09-22:
@Bostwickenator Alexander, any chance you could take a look, willing to pay $$ for that :)

Razikus, 2024-09-23:
Hi
I would also be very interested into supporting of nano

blackkarlsson, 2024-09-24:
@Razikus as a developer or supporter? ;)

anyway I'm very open to support financially this issue :) please pm me if someone open for that

Razikus, 2024-09-24:
haha - I can develop it also, but I don't have yet my Nano camera, therefore I'm looking into existing solutions

Im not sure yet which camera will I choose, but I will let you know when this will happen

Bostwickenator, 2024-09-24:
We aren't accepting financial support. I'll be investigating feasibility of
supporting the Nano.

blackkarlsson, 2024-09-24:
thanks a lot in advance, Alex!

blackkarlsson, 2024-09-25:
@Bostwickenator if it would be helpful I can amazon one Nano 300 to you, let me know if needed

Bostwickenator, 2024-09-25:
@blackkarlsson thanks but no. If supporting these cameras is possible it
will take a while. As a side note if you need something immediately I
suggest sourcing a Seek Compact Pro or Infiray P2.

blackkarlsson, 2024-09-25:
I know that it might take a while, that's the reason I suggested custom development :)

P2Pro is a good one, but Nano300 has better sensor resolution(320x240 vs 256x192), so that's why I choose to test it. 

it would be great if you could at least dig into how time-consuming this issue might be to have at least some insights what to do next. thanks in advance!

waayaa, 2024-10-28:
@blackkarlsson any update on this? About to embark on integrating Nano with motion.
cc: @Bostwickenator

icb-, 2024-12-20:
Would a USB capture between an iPhone and Nano 300 be helpful, or do you have everything you need (except time)?

Bostwickenator, 2024-12-21:
I have the Nano 300 and its performance is poor. Simply put I am not going
to invest time in it when the cameras from Chinese manufacturers are
cheaper, better, and not hobbled by export controls.

I realize this may disappoint people.

Seek does furnish an SDK for their cores if you absolutely have to use
their hardware.